STMicroelectronics/STM32CubeL1 #2: Duplicate defines when using CMSIS 5 headers

salkinium, 2019-06-05:
**Describe the set-up**
 * arm-none-eabi-gcc
 * Using `core_cm3.h` header from CMSIS_5 repository

**Describe the bug**

The device headers add some missing Cortex-M peripheral defines for `SCB_CFSR_*`, which were missing in the CMSIS 4 headers.

In CMSIS 5, however, these defines have been added, thus resulting in a warning:
```
warning: "SCB_CFSR_*_Pos" redefined
warning: "SCB_CFSR_*_Msk" redefined
```

We use the same CMSIS 5 files for all STM32 families, so we can guarantee feature consistency and receive updates (CMSIS 4 is EoL). So this isn't a bug for you (yet), just for me.

**How To Reproduce**

Upgrade the CMSIS 4 files to CMSIS 5 and compile.

**Additional context**

There is a backwards compatible fix, by checking the major CMSIS version (`__CM3_CMSIS_VERSION_MAIN`). This fixes my issue and should work for you too.
See [full patch](https://github.com/modm-io/cmsis-header-stm32/blob/master/patches/l1-2.patch).

salkinium, 2019-06-05:
The [CMSIS 4 definitions are here](https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/core_cm3.h#L595-L603).
Compare with the [CMSIS 5 definitions here](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/core_cm3.h#L552-L614).

CCASTM, 2019-06-05:
Hello Niklas,
Can you indicate which version of the STM32CubeL1 firmware package you are referring to ?
Regards
Christophe

salkinium, 2019-06-05:
This refers to the CMSIS header files inside the latest `STM32CubeL1.zip` file downloaded [from your homepage](https://www.st.com/en/embedded-software/stm32cubel1.html).
Checking `STM32Cube_FW_L1_V1.9.0/Drivers/CMSIS/Device/ST/STM32L1xx/Include/stm32l151xe.h` for example yields (line 8803):
```c
#define SCB_CFSR_BFARVALID_Pos            (SCB_CFSR_BUSFAULTSR_Pos + 7U)       /*!< SCB CFSR (BFSR): BFARVALID Position */
#define SCB_CFSR_BFARVALID_Msk            (1UL << SCB_CFSR_BFARVALID_Pos)      /*!< SCB CFSR (BFSR): BFARVALID Mask */
#define SCB_CFSR_BFARVALID                  SCB_CFSR_BFARVALID_Msk             /*!< Bus Fault Address Register address valid flag */
```
(None of the other `SCB_*` defines in that header define the `_Msk` or `_Pos` values, it's only `SCB_CFSR_*`).

Correction: you are already using CMSIS 5, so you should already see this issue yourself.
In `STM32Cube_FW_L1_V1.9.0/Drivers/CMSIS/Core/Include/core_cm3.h` line 580:
```c
/* BusFault Status Register (part of SCB Configurable Fault Status Register) */
#define SCB_CFSR_BFARVALID_Pos            (SCB_CFSR_BUSFAULTSR_Pos + 7U)                  /*!< SCB CFSR (BFSR): BFARVALID Position */
#define SCB_CFSR_BFARVALID_Msk            (1UL << SCB_CFSR_BFARVALID_Pos)                 /*!< SCB CFSR (BFSR): BFARVALID Mask */
```

salkinium, 2019-06-05:
So since your using CMSIS 5 for L1 too, my patch should just remove all `SCB_CFSR_*_Pos` and `SCB_CFSR_*_Msk` defines, instead of keeping them there for CMSIS 4 compatiblity. That makes it even simpler, yay!

CCASTM, 2019-06-06:
OK, it seems that we have identified this point but forgot to fully update the cmsis device file
If confirmed by the development team, you can keep your correction, and we will correct our next delivery
Regards
Christophe

ASELSTM, 2019-06-20:
ST Internal Reference: 68405

CCASTM, 2019-08-21:
Hello Niklas,
Your correction proposal is valid.
We have integrated the correction in our software database. It will be available in our next STM32L1 publication.
If you have no other question, can we close this Issue ?
Regards
Christophe

closed 2019-08-21 by salkinium