STMicroelectronics/STM32CubeL1 #6: FLASH_SIZE incorrectly reports flash size for cat.2 devices with DEV_ID 0x429

Xeenych, 2020-10-30:
FLASH_SIZE incorrectly reports flash size for cat.2 devices with DEV_ID 0x429

```
smt32lxx_hal_flash.h
#define FLASH_SIZE                (uint32_t)((*((uint32_t *)FLASHSIZE_BASE)&0xFFFFU) * 1024U)
```
```
From Reference Manual
Note: For DEV_ID = 0x429, only LSB part of F_SIZE: F_SIZE[7:0] is valid. The MSB part 
F_SIZE[15:8] is reserved and must be ignored.
```

RKOUSTM, 2020-11-17:
Hi @Xeenych,

Thank you for your contribution. The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

RKOUSTM, 2020-11-17:
ST Internal Reference: 96492

RKOUSTM, 2020-11-18:
Hi @Xeenych,

Thank you for your contribution. A fix for this code below :
https://github.com/STMicroelectronics/STM32CubeL1/blob/c4349cf67f149b15ae9bb44b00e5a48c76361bfb/Drivers/STM32L1xx_HAL_Driver/Inc/stm32l1xx_hal_flash.h#L124

will be published soon in a future release (STM32L1Cube V1.10.3) looks like the following:
```diff
+ #if defined (FLASH_CUT1) || defined (FLASH_CUT2)
+ #define FLASH_SIZE                      (uint32_t)((*((uint32_t *)FLASHSIZE_BASE)&0xFFU) * 1024U)
+ #else /*FLASH_CUT3 || FLASH_CUT4 || FLASH_CUT5 || FLASH_CUT6*/
+ #define FLASH_SIZE                      (uint32_t)((*((uint32_t *)FLASHSIZE_BASE)&0xFFFFU) * 1024U)
+ #endif
```
In the CMSIS files, a specific definition for all device categories (Cat.1, Cat.2, Cat.3, Cat.4, Cat.5 and Cat.6) is added as follows for Cat.2 devices :
```diff
+  /*
+   * @brief Specific device feature definitions (not present on all devices in the STM32L1 serie)
+   */
+ #define FLASH_CUT2
```
Thank you again for your contribution.

With regards,

closed 2021-06-08 by RKOUSTM