STMicroelectronics/STM32CubeL1 #7: Missing linkerfiles templates for gcc

stefaneicher, 2021-02-22:
Hi,

did you intentionally not put the linkerfiles in to this repo for gcc`?

https://github.com/STMicroelectronics/STM32CubeL1/tree/master/Drivers/CMSIS/Device/ST/STM32L1xx/Source/Templates/gcc

Regards Stefan

RKOUSTM, 2021-02-23:
Hi @stefaneicher ,

Allow me first to thank you for your contribution. Your report has been forwarded to our development team. We will get back to you as soon as we have more details.

With regards,

stefaneicher, 2021-02-25:
ok thanks.
The same is actually the case in cmsis_device_l1 https://github.com/STMicroelectronics/cmsis_device_l1/tree/master/Source/Templates/gcc

stefaneicher, 2021-02-25:
..and in  cmsis_device_f0
https://github.com/STMicroelectronics/cmsis_device_f0/tree/master/Source/Templates/gcc
and STM32CubeF0
https://github.com/STMicroelectronics/STM32CubeF0/tree/master/Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/gcc

RKOUSTM, 2021-02-25:
Hi @stefaneicher,

Thank you for this report. It has been forwarded to our development teams. I will get to you as soon as they provide their feedback.

With regards,

RKOUSTM, 2021-05-08:
Hi @stefaneicher,

First, allow me to thank you for your contribution. Indeed, the GCC linker files has only been made for recent products such as H7, WB/WL, L5... because these series have brought some specificities (Dual core, secure / non secure portioning). However, no plain for our technical team to add GCC linker files for legacy products (L1, F0, F1...).  

Please, allow me to close this issue. Thank you once again for your contribution and thank you for your comprehension.

With regards,

closed 2021-05-08 by RKOUSTM