STMicroelectronics/STM32CubeL1 #11: FW_L1 V1.10.3 does not set VTOR in SystemInit() by default, which causes HardFault.

popons, 2021-10-05:
**Caution**
In v1.10.3, SCB->VTOR is not set by default and remains at 0x00000000.
Nevertheless, g_pfnVectors will be placed at 0x08000000.
So if you run the default project in STM32CubeIDE, you will get a HardFault from SysTick Exception.

**Describe the set-up**
 * Custom board with STM32L151CB
 * STM32CubeIDE and FW_L1 V1.10.3

**Describe the bug**
In the project's default configuration, USER_VECT_TAB_ADDRESS is not defined and VTOR remains at 0x00000000, resulting in a HardFault.
Is this intended?

**How To Reproduce**
1. Open STM32CubeIDE 1.6.1

2. Create New STM32 Project

3. Select STM32L151CB-A

4. Select V1.10.3 Firmware

5. Build and Program and Run

**Additional context**

**Screenshots**[image]

popons, 2021-11-02:
Thank you for your reply.
I checked and BOOT0 was in OPEN state.
I shorted it to GND and it booted normally.
Thank you for your time responding to my inquiries.

ALABSTM, 2021-11-05:
Hi @popons,

Glad to know you could solve this point. Please allow me to close this issue in this case.

With regards,

closed 2021-11-05 by ALABSTM