STMicroelectronics/STM32CubeL1 #12: stm32l151xc.h lacks definitions for channel 27 to 29 in ADC sample time register 1

celange, 2021-11-02:
**Describe the set-up**
 * STM32CubeL1 v1.10.3

**Describe the bug**
* **stm32l151xc.h** lacks definitions for channel 27, 28 and 29 in the ADC sample time register 1 (ADC_SMPR1)

**How To Reproduce**
1. Get STM32CubeL1 v1.10.3.

2. Inspect section **Bit definition for ADC_SMPR1 register** in **stm32l151xc.h** starting at line 986.
    There are definitions for channel 20 to channel 26 only, channel 27, 28 an 29 are missing.

3. Check RM0038 Reference Manual chapter 12.15.4 ADC sample time register 1 (ADC_SMPR1)
    The sample time can be configured for channel 20 to channel 29.

RKOUSTM, 2021-11-15:
Hi @celange,

Thank you for your report. According to the datasheet [DS8890](https://www.st.com/content/ccc/resource/technical/document/datasheet/2a/6e/97/91/cd/c0/43/8b/DM00048356.pdf/files/DM00048356.pdf/jcr:content/translations/en.DM00048356.pdf), the STM32151xC has 25 channels. While the STM32L151xCA, STM32L151xD, STM32L152xCA, STM32L152xD, STM32L162xCA, STM32L162xD, STM32L151xE, STM32L151xDX, STM32L152xE, STM32L152xDX, STM32L162xE and STM32L162xDX have up to 40 channels.

However, the channel numbers are still inadequate compared to those mentioned in the CMSIS files for each device. Your report will be forwarded to our development team. We will get back to you as soon as we have more details.

Thank you again for your report. 

With regards,

RKOUSTM, 2021-12-08:
ST Internal Reference: 119154

HBOSTM, 2023-04-28:
Hello @celange,

The fix has been implemented and is now available in the frame of the latest STM32CubeL1 package V1.10.4 release. This issue can be closed now. Thank you again for your contribution.

With Regards,

closed 2023-04-28 by HBOSTM