STMicroelectronics/STM32CubeL1 #13: Preprocessor directives placed by mistake in "stm32l1xx_ll_spi.h"

StressTest, 2021-12-16:
`#if defined(SPI_CR2_FRF)
/**
  * @brief  Set serial protocol used. TI Mode not supported for Category 1 and 2.
  * @note   This bit should be written only when SPI is disabled (SPE = 0) for correct operation.
  * @rmtoll CR2          FRF           LL_SPI_SetStandard
  * @param  SPIx SPI Instance
  * @param  Standard This parameter can be one of the following values:
  *         @arg @ref LL_SPI_PROTOCOL_MOTOROLA
  *         @arg @ref LL_SPI_PROTOCOL_TI
  * @retval None
  */
__STATIC_INLINE void LL_SPI_SetStandard(SPI_TypeDef *SPIx, uint32_t Standard)
{
  MODIFY_REG(SPIx->CR2, SPI_CR2_FRF, Standard);
}

/**
  * @brief  Get serial protocol used
  * @rmtoll CR2          FRF           LL_SPI_GetStandard
  * @param  SPIx SPI Instance
  * @retval Returned value can be one of the following values:
  *         @arg @ref LL_SPI_PROTOCOL_MOTOROLA
  *         @arg @ref LL_SPI_PROTOCOL_TI
  */
__STATIC_INLINE uint32_t LL_SPI_GetStandard(SPI_TypeDef *SPIx)
{
  return (uint32_t)(READ_BIT(SPIx->CR2, SPI_CR2_FRF));
}
#endif
`
No need processor brackets **#if defined(SPI_CR2_FRF) ... #endif**

ASELSTM, 2022-01-07:
Hi @StressTest,

As mentioned in the reference manual [RM0038](https://www.st.com/resource/en/reference_manual/rm0038-stm32l100xx-stm32l151xx-stm32l152xx-and-stm32l162xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf),  the `SPI_CR2_FRF` bit represents the SPI TI mode which is only available in the category 3, category 4, category 5 and category 6 devices. 

[image]

The  devices categories are described in the **"section 1.5 Product category definition"** of the reference manual. The STM32L100RB is a category 1 device as mentioned in the table below .

[image]

Therefore, the SPI TI mode, and consequently the `SPI_CR2_FRF` bit, is not available in the STM32L100RB part number. Thus, there is no bug in the CMSIS file, the issue is rather related to the CubeMX generation problem as the code generated by CubeMx, is calling the `LL_SPI_SetStandard` function which is managed by the `SPI_CR2_FRF` bit.

Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories.  They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the **ST Community** . 

Please allow me then to close this thread. Thank you for your comprehension.

closed 2022-01-07 by ASELSTM

ASELSTM, 2022-01-07:
ST Internal Reference: 120476