STMicroelectronics/STM32CubeL1 #17: SystemCoreClockUpdate() does not work correctly when SYSCLK source is HSI.

yn386, 2022-09-05:
**Describe the set-up**
 * The board: NUCLEO-L152RE

**Describe the bug**
SystemCoreClockUpdate() does not work correctly when SYSCLK source is HSI.

If SYSCLK source is HSI and HSI_VALUE is not defined, SystemCoreClockUpdate() cannot update SystemCoreClock correctly.

According to the STM32L1 microcontorollers specification, the HSI value is 16MHz.
Also, the comment of SystemCoreClockUpdate() describes that the default HSI_VALUE is 16MHz.
https://github.com/STMicroelectronics/STM32CubeL1/blob/5641cb07a1d78d7e9649d056136e513bb273fcca/Drivers/CMSIS/Device/ST/STM32L1xx/Source/Templates/system_stm32l1xx.c#L196-L198
However, default HSI_VALUE is defined as 8MHz at
https://github.com/STMicroelectronics/STM32CubeL1/blob/5641cb07a1d78d7e9649d056136e513bb273fcca/Drivers/CMSIS/Device/ST/STM32L1xx/Source/Templates/system_stm32l1xx.c#L69-L72

HSI_VALUE can be adapted by the user appliication, but  I think default HSI_VALUE should be 16MHz.

**How To Reproduce**
HSI_VALUE is not defined.
To get 32MHz HCLK, I set
- Use HSI
- PLLMul x6
- PLLDiv /3

After setting these parameters, call SystemCoreClockUpdate().
Global variable 'SystemCoreClock' expected 32MHz, but I got 16MHz.

**Additional context**
I changed default HSI_VALUE as follows.
```
--- a/CMSIS/STM32L1xx/Source/Templates/system_stm32l1xx.c
+++ b/CMSIS/STM32L1xx/Source/Templates/system_stm32l1xx.c
@@ -67,7 +67,7 @@
 #endif /* HSE_VALUE */
 
 #if !defined  (HSI_VALUE)
-  #define HSI_VALUE    ((uint32_t)8000000U) /*!< Default value of the Internal oscillator in Hz.
+  #define HSI_VALUE    ((uint32_t)16000000U) /*!< Default value of the Internal oscillator in Hz.
                                                 This value can be provided and adapted by the user application. */
 #endif /* HSI_VALUE */

```

ASELSTM, 2022-09-08:
ST Internal Reference: 134336

HBOSTM, 2023-04-28:
Hello @yn386 ,

The fix has been implemented and is now available in the frame of the latest STM32CubeL1 package V1.10.4 release. This issue can be closed now. Thank you again for your contribution.

With Regards,

closed 2023-04-28 by HBOSTM